Inteform/esp32-lora-library #2: Check for value of version register fails after lora_reset (in lora_init).

vijaygill, 2018-08-25:
I tried this library on two boards I have here and adding a small delay fixed this.

Tested it again many times by multiple resets of the board (button / power recycle) and no issues so far. The assert "assert(version == 0x12);" fails sometimes after a reset, but after the board reboots, everything is back to normal. Library seems quite stable.

I will raise PR.

bruabas, 2018-08-27:
Hi Gill,
Thank you very much. Indeed, those kinds of timing issues are the worst part...
I will include your delay with a small change.
Are you using the library in practical projects? Notice that this library still misses the frequency hopping implementation, I am working on it yet. 
Thank you very much for your help!

closed 2018-08-27 by bruabas

vijaygill, 2018-08-27:
Hi @bruabas,

Thanks for your reply. I am just beginning to use LoRa on two esp32 boards I had lying around. No project yet, but soon it will become part of a mesh on motio  sensing devices in a large farm.

eric-inteform, 2018-08-27:
OK. Which country are you from?

----------------------------------------------------------------------------------------
INTEFORM
Tel: +55 11 3227-5189;  Whatsapp: 11 952897905
Av Ipiranga 1100, Sala 90/94, Centro, Sao Paulo, CEP 01040-904
Email: vendas@inteform.com.br <Eric.Shen@inteform.com.br>*, *Site:
www.inteform.com.br
Facebook: https://www.facebook.com/inteformTec/
INTEFORM TECNOLOGIA E COMERCIO EIRELI CNPJ: 22.779.632/0001-38

vijaygill, 2018-08-27:
Republic of Ireland

PR #3 (2018-08-27): Added a small delay after reset or else it caused assert to fail.
Fixes #2.