Inteform/esp32-lora-library #5: Deep sleep issue

imelekhin, 2018-08-29:
Hi!
It looks like lora_init() does not work correctly after  esp32 deep sleep wakeup.
here is debug output:
assertion "version == 0x12" failed: file "/home/iho/esp/hello_world/components/lora/lora.c", line 357, function: lora_init
abort() was called at PC 0x400d295b on core 0
part of code to reproduce:

```
void app_main()
{

	esp_sleep_wakeup_cause_t cause = esp_sleep_get_wakeup_cause();

	lora_init();
	lora_set_frequency(433e6);
	lora_enable_crc();

	/*while(1){
		lorasend();
		vTaskDelay(2000/portTICK_PERIOD_MS);
	}*/

    if (cause != ESP_SLEEP_WAKEUP_ULP) {
	  //      printf("Not ULP wakeup\n");

            init_ulp_program();
	    } else {
	  //      printf("Deep sleep wakeup\n");
	        proceed_buffer();
	        lorasend();
	        for(int i=0;i<FEATURES_NUM;i++){
	        			features[i] = features_calc[i]();
	        		}

	    }
	    //printf("Entering deep sleep\n\n");
	    start_ulp_program();
	    ESP_ERROR_CHECK( esp_sleep_enable_ulp_wakeup() );
	    esp_deep_sleep_start();

}
```

in while(1){} all works fine, when we go to deep sleep first execution (ESP_SLEEP_WAKEUP_ULP) works fine, all folowing (i.e. after ULP wake-up generate assertion blah-blah....

imelekhin, 2018-08-30:
Brubas, adding vTaskDelay(100) does not help. Programm fails to assert() and do a reboot. As a workaround, I commented assert() - everything works fine,,,,,

bruabas, 2018-08-30:
This is really strange. Let's check... 
Anyway, you do not need to reinit LoRa after exiting deep-sleep, if the transceiver power is not turned off.