Inteform/esp32-lora-library #13: LoRa sender not working

lucas-figueiredo-m, 2019-09-01:
Hello everyone! Yesterday I had cloned the repo, followed the instructions on the README and created 2 esp32 projects on my PC: one for the sender and another for the receiver. But it seems that the sender code isn't working.

Triyng to debug the code, I have inserted some printf functions inside the lora_send_packet function on lora.c . I have found that the code isn't escaping the following 
while loop:

`while((lora_read_reg(REG_IRQ_FLAGS) & IRQ_TX_DONE_MASK) == 0) {
      vTaskDelay(2);
      printf("Waiting\n");
   }`

Do you have faced the same problem? Are those regAddresses right? I think that I'm doing something wrong, but I can't see what.

For instance: I'm using the SX1278 chip ( the RA-01 module). It can only uses 433MHz as carrier for the transmission

AloyseTech, 2019-11-07:
I currently have the same issue. Have been able to solve this? I'm starting to investigate.

eduardsui, 2020-01-16:
Same here, tested with SX1278 and LoPy board. It gets stuck in the same loop. Any news about this?

eduardsui, 2020-01-16:
Working now. For me it was the RST pin. Connected to EN and worked perfectly afterwards.

krupis, 2023-07-30:
I have simmillar issue. I normally find this issue when turning on my device for the first time, LoRa module will get stuck in this while loop for some time (sometimes around a minute or so) and then will start working without any issues as normal. I have created a seperate issue for this:
https://github.com/Inteform/esp32-lora-library/issues/26

I dont see how RST pin can have anything to do with this. Did you investigate it further and found anything else?