Inteform/esp32-lora-library #18: Does not work on the newer versions of ESP-IDF had to go back to  3.3.2

ambassallo, 2021-02-13:


ephro, 2021-07-09:
I had the same problem. 

Move lora.c into a new directtory called `src` then add a CMakeLists.txt with the following

```
set(COMPONENT_SRCDIRS "src")
set(COMPONENT_ADD_INCLUDEDIRS "include")
register_component()
```

pedrominatel, 2021-07-19:
Hi @bruabas, this is very easy to fix. Do you have plans to fix or can I PR the solution?

nopnop2002, 2022-02-03:
Try this:
https://github.com/nopnop2002/esp-idf-sx127x

pedrominatel, 2022-02-07:
Thank you @pedrominatel! Do you consider adding your version to our [Components Manager](https://components.espressif.com/)?

nopnop2002, 2022-02-07:
@pedrominatel 

I don't understand the relationship between Components Manager and the github repository.

Are these in sync?

If I update the github repository, will the contents of Components Manager be updated as well?

pedrominatel, 2022-02-08:
Hi @nopnop2002, yes, they are in sync and the only thing you need to do is to add one more file to your repo.

See this as an example on [littlefs](https://github.com/joltwallet/esp_littlefs) and the required file example [idf_component.yml](https://github.com/joltwallet/esp_littlefs/blob/master/idf_component.yml)

nopnop2002, 2022-02-08:
@pedrominatel 

Thank you for the explanation.
I will try.