Inteform/esp32-lora-library #26: Sometimes device stuck in while ((lora_read_reg(REG_IRQ_FLAGS) & IRQ_TX_DONE_MASK) == 0)

krupis, 2023-07-30:
Hello. I have noticed an issue where my device is stuck when trying to send lora packet. I have added some extra printf statements in lora_send_packet() function:

```
void lora_send_packet(uint8_t *buf, int size)
{
   /*
    * Transfer data to radio.
    */
   lora_idle();
   lora_write_reg(REG_FIFO_ADDR_PTR, 0);

   for (int i = 0; i < size; i++)
      lora_write_reg(REG_FIFO, *buf++);

   lora_write_reg(REG_PAYLOAD_LENGTH, size);

   /*
    * Start transmission and wait for conclusion.
    */
   printf("Starting transmission \n");
   lora_write_reg(REG_OP_MODE, MODE_LONG_RANGE_MODE | MODE_TX);
   while ((lora_read_reg(REG_IRQ_FLAGS) & IRQ_TX_DONE_MASK) == 0)
      vTaskDelay(2);

   lora_write_reg(REG_IRQ_FLAGS, IRQ_TX_DONE_MASK);
   printf("Transmission done \n");
}
```

and from the serial logs I see:

```
[1B][0;32mI (0) cpu_start: Starting scheduler on APP CPU.[1B][0m
Initializing lora 
[1B][0;32mI (313) gpio: GPIO[20]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [1B][0m
[1B][0;32mI (323) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [1B][0m
Setting frequency 
Enablig crc 
create lora tx task 
Starting transmission 
56: Transmission done 
packet sent...
01:06: Starting transmission 
Transmission done 
packet sent...
01:16: Starting transmission 
Transmission done 
packet sent...
01:26: Starting transmission 
Transmission done 
packet sent...
01:36: Starting transmission 
```

I have enabled timestamps in the serial terminal (Termite) and as you can see after the device boots up, first transmission is completed  56 seconds after the device start. After that, it transmits periodically after every 10 seconds as it is supposed to.

It almost feels like the LoRa module needs to "warmup" before it can send the message.

Did anyone else had this issue? And could help me figure out what could be the cause?

Its very important to me as I do not want my device to stay on for 56 seconds unnecessary since I am working on battery powered lora device, I need my device to send the message ASAP and go to sleep.

nopnop2002, 2024-07-08:
Requires device initialization rather than warm-up.

adityanijave, 2024-07-12:
> Hello, first of all I want to say thanks for providing this easy to use lora library. However, I have noticed an issue where my device is stuck when trying to send lora packet. I have added some extra printf statements in lora_send_packet() function:
> 
> ```
> void lora_send_packet(uint8_t *buf, int size)
> {
>    /*
>     * Transfer data to radio.
>     */
>    lora_idle();
>    lora_write_reg(REG_FIFO_ADDR_PTR, 0);
> 
>    for (int i = 0; i < size; i++)
>       lora_write_reg(REG_FIFO, *buf++);
> 
>    lora_write_reg(REG_PAYLOAD_LENGTH, size);
> 
>    /*
>     * Start transmission and wait for conclusion.
>     */
>    printf("Starting transmission \n");
>    lora_write_reg(REG_OP_MODE, MODE_LONG_RANGE_MODE | MODE_TX);
>    while ((lora_read_reg(REG_IRQ_FLAGS) & IRQ_TX_DONE_MASK) == 0)
>       vTaskDelay(2);
> 
>    lora_write_reg(REG_IRQ_FLAGS, IRQ_TX_DONE_MASK);
>    printf("Transmission done \n");
> }
> ```
> 
> and from the serial logs I see:
> 
> ```
> [1B][0;32mI (0) cpu_start: Starting scheduler on APP CPU.[1B][0m
> Initializing lora 
> [1B][0;32mI (313) gpio: GPIO[20]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [1B][0m
> [1B][0;32mI (323) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [1B][0m
> Setting frequency 
> Enablig crc 
> create lora tx task 
> Starting transmission 
> 56: Transmission done 
> packet sent...
> 01:06: Starting transmission 
> Transmission done 
> packet sent...
> 01:16: Starting transmission 
> Transmission done 
> packet sent...
> 01:26: Starting transmission 
> Transmission done 
> packet sent...
> 01:36: Starting transmission 
> ```
> 
> I have enabled timestamps in the serial terminal (Termite) and as you can see after the device boots up, first transmission is completed 56 seconds after the device start. After that, it transmits periodically after every 10 seconds as it is supposed to.
> 
> It almost feels like the LoRa module needs to "warmup" before it can send the message.
> 
> Did anyone else had this issue? And could help me figure out what could be the cause?
> 
> Its very important to me as I do not want my device to stay on for 56 seconds unnecessary since I am working on battery powered lora device, I need my device to send the message ASAP and go to sleep.

yes I am facing the same issue